rosflight/rosflight_ros_pkgs #8: ROS release

dpkoch, 2017-05-18:
Need to figure out how to release this stack through the ROS build farm.

dpkoch, 2017-05-18:
Also need to remember to bump version tags in all `package.xml` files

dpkoch, 2017-05-24:
Closed by ros/rosdistro#15049

dpkoch, 2017-05-24:
That last release failed on the build farm due to some missing dependencies; I think I've fixed that, should be resolved by ros/rosdistro#15054

superjax, 2017-05-24:
Cool!  So, how does this work?  I understand that you somehow made `rosflight-release` a part of `rosdistro` but how does that make its way to the main repository?

dpkoch, 2017-05-24:
One more try: ros/rosdistro#15057

@superjax Magic.

So I think once they merge the pull request it triggers a new build on their Jenkins buildfarm server. The buildfarm uses the `rosdistro` configuration file to know which packages to include and what version, so the info we add to that file tells the buildfarm to include our packages. Then every once in a while those built versions get pulled into the debian repository; still not positive how/when that step happens. It's magic.

dpkoch, 2017-06-03:
See ros/rosdistro#15149

dpkoch, 2017-06-05:
Closed by ros/rosdistro#15149

closed 2017-06-05 by dpkoch