rosflight/rosflight_ros_pkgs #20: Rosflight Baro measurement calibration

dallinbriggs, 2017-07-12:
This data is pulled from a recent flight of rosflight where two other state estimators were also on-board. As you can see, the measured pressure is different than the other two pressure measurements. The INS and the Pixhawk both use GPS to estimate the barometer bias, but rosflight does not. This means that when you calibrate the altitude at the beginning of the flight, it will indeed set your altitude to zero, but your altitude estimate will not be accurate as you climb in elevation. I observed about a 10 meter altitude difference when the aircraft was at an altitude of 70 meters.
[image]

superjax, 2017-07-20:
This I believe should be fixed by the new way that the firmware calibrates IMU

superjax, 2017-07-31:
This was fixed by @len0rd

closed 2017-07-31 by superjax