rosflight/rosflight_ros_pkgs #24: SIL Error with calibrating IMU

gellings, 2017-07-19:
When calibrating the imu for rosflight_sim fixedwing the autopilot reports an error: _[Autopilot]: Too much movement for IMU cal_  

Any thoughts

gellings, 2017-07-19:
I just updated to the latest rc1.0 and it seems the error went away but I still can't get it to arm on fixed-wing sil

gellings, 2017-07-19:
I tried plotting the imu but there seems to be another bug with the imu time-stamp.  The imu does seem to be a little bit wild.  I could be because the aircraft interaction with the ground is weird.  Adding wheels to the vehicle seemed to not make a difference.

superjax, 2017-07-20:
I fixed the IMU time stamp bug, but I still don't know what is going on with the IMU calibration.  I would try setting the noise to zero and see if it still worked, or trying to print out the max - min vector when the IMU calibration is performed.

commit b26ef173 (2017-07-28): fixed #24
closed 2017-07-31 by superjax