rosflight/rosflight_ros_pkgs #31: Negative Airspeed readings

dallinbriggs, 2017-07-21:
It seems as though the differential pressure sensor can return outliers for the pressure reading. This causes the airspeed to go negative for a second. We should probably do some outlier rejection on this. 
[image]

drinkdhmo, 2017-07-26:
We should implement outlier rejection or other filtering at the ROScopter/ROSplane level so that high level applications get raw data from the sensors and can then best decide how to filter it.

superjax, 2017-07-28:
Does anyone want to implement this?  (It would be best in the sensors.cpp file of the firmware)

superjax, 2018-02-28:
This was fixed a long time ago!  Thanks @drinkdhmo

closed 2018-02-28 by superjax