rosflight/rosflight_ros_pkgs #32: Crazy servo output when aircraft is disarmed

dallinbriggs, 2017-07-21:
I think something may be wrong with the servo output while sitting disarmed. Having a high pwm update rate causes the servo output to be completely random. In this plot, all I did was switch the update rate from the default frequency to 50 Hz for servos. 

[image]

superjax, 2017-07-21:
@dallinbriggs, can you post your parameters yaml file here?

dpkoch, 2017-07-21:
@dallinbriggs It appears that this behavior only occurs when the autopilot is in failsafe mode. This happens because the failsafe command is zero setpoints in angle mode, so the attitude controller is running. Can you confirm that this stops happening if you have RC on?

dpkoch, 2017-07-21:
Replaced by rosflight/firmware#217

closed 2017-07-21 by dpkoch