rosflight/rosflight_ros_pkgs #33: Coordinate Frames

plusk01, 2017-07-22:
What coordinate frame standard are we adopting for the ROSflight stack? It would be good to have a standard and have some solid documentation (with images) on this. Coordinate frames are a big source of confusion in robotics it seems.

This comes with some philosophical questions: ROS standard ([REP 103](http://www.ros.org/reps/rep-0103.html) and [REP 105](http://www.ros.org/reps/rep-0105.html)) is ENU with body-NWU, which lends itself to ground robots. But the UAV community is NED.

If we break the ROS standard and use NED, it may make it more difficult/confusing to interface with other ROS packages. Also, for what it is worth, `mavros` follows the ROS standard, but didn't really seem to agree on it until well into the project. And even then, they seem to have had some errors in their understanding: https://github.com/mavlink/mavros/issues/216

It may also be worth considering `tf2` to help mitigate coordinate frame issues.

plusk01, 2017-07-22:
Related: https://github.com/byu-magicc/rosflight_plugins/issues/4

superjax, 2017-07-28:
From a practical perspective, I'm pretty determined to continue breaking REP.  Haha.  The aerospace literature in general follows a NED convention.  ROS's origins are in ground robots, which use ENU.   I don't know that it's super unreasonable to stick with a NED convention for MAVs.

I know several people who have been incredibly confused by the switching of coordinate frames in mavros.  I think that's probably the worst possible implementation (a hidden coordinate frame switch?)

Anyway.  I'm not super familiar with tf.  Do you have any idea on how we might deal with that?   In my implementation of `rosflight_sil` I publish truth in both NED and NWU to appeal to both groups.  They are namespaced in the publication to make it obvious which frame they are in.  Perhaps tf would make that easier.

superjax, 2018-11-29:
Let's publish the attitude to tf.  I think that would be a nice and easy way to visualize the attitude.