rosflight/rosflight_ros_pkgs #34: Sensor calibration in presence of bad measurements

drinkdhmo, 2017-07-26:
The current scheme of sensor calibration that averages the first 100 samples from a sensor is susceptible to problems when a sensor outputs a few grossly bad measurements. This can pull the calibration way off.

I suggest we add some intelligence to the calibration to check if the samples collected are reasonable. For example, we could calculate the sample variance and reject the calibration if it is above some threshold.

superjax, 2017-07-28:
We are actually doing this with the IMU already, just not with the baro or airspeed.  As a pretty easy hack, I just check the max and min values, and if they are super far apart, then I say that the calibration didn't work.  
[sensors.cpp #L353](https://github.com/rosflight/firmware/blob/6d67e928a739d8e5d4eb0dd4a36dee1c9f1a4bad/src/sensors.cpp#L353)

superjax, 2018-02-28:
Thanks @drinkdhmo for fixing this.

closed 2018-02-28 by superjax