rosflight/rosflight_ros_pkgs #36: Plane Doesn't Fly

superjax, 2017-07-28:
So I fixed #24, and we can now fly the plane.  However, I'm pretty sure something is wrong with the dynamics, because when I get enough airspeed that we start to run the aerodynamics, the plane goes into a flat spin.  It's sort of amusing to watch, but it's not actually what I want. Haha.

superjax, 2017-07-28:
I tried adding wheels per @gellings suggestion.  Still didn't work.  I suspect that something is wrong in the dynamics.  Perhaps later, I'll compare them with the book.

gellings, 2017-07-31:
I just looked over the dynamics and I haven't found anything yet.  I'll keep looking at them if we are sure they are the problem.  Are we sure that its not a problem with gazebo, we did have to do a funny rotation to make the module line up right. Could the dynamics be calculated in one direction and applied in another?

gellings, 2017-07-31:
see PR #37, it was a bug introduced to the dynamics after all.

closed 2017-07-31 by gellings