rosflight/rosflight_ros_pkgs #38: Launch Files are Kinda Useless

superjax, 2017-07-31:
So, we were talking, and while the default launch files are minimal and that's nice, they are actually a little useless, because you need an RC simulation node and `rosflight_io` to get anything done. 

So discussion:  Is it worth adding launch files that we can use to actually fly the aircraft in sim with?

superjax, 2017-07-31:
Some mentions so I'm not just discussing this with myself:

@dpkoch @gellings

gellings, 2017-07-31:
My opinion, right now, is that gazebo sil (fw/mr), rosflight_io, and rc_joy should all be launched from the same file.  It might make since to have rc_joy and the launch file in rosflight_utils.

gellings, 2017-07-31:
I just pushed an example of what I'm thinking of for fixedwing SHAd1a22bff741122730ca425e361fe70546d966d0a 
I probably should put it in a branch, but since we haven't got around to cleaning our rosflight_utils, I didn't worry about it.

gellings, 2017-07-31:
We also have about 6 layers of abstraction. In this case I think the abstraction leads to confusion rather then clarity.  For example, a fixedwing simulation requires; fixedwing_sil.launch, fixedwing.launch, base.launch, spawn_mav.launch, fixedwing.urdf.xacro, and rosflight_sil.xacro.  Its really only 3 nodes, an xacro and some params.  It shouldn't need this many layers.

gellings, 2017-07-31:
They are also useless without a GPS plugin.  Unless maybe your interested in trying to solve the GPS-denied problem too...   Could be solved by adding a GPS driver to the firmware and then added the simulated sensor.

dpkoch, 2017-08-01:
One argument against including everything in the launch files: From a user perspective it would be nice to have my own launch file that I could run on either hardware or simulations. So it would launch rosflight_io and everything either way, but then I set a sim flag and it optionally launches the simulation to replace hardware. For that use case, it's nice to have the sim launch file be as minimal as possible. I'm not saying the way we have it set up right now is the best or even particularly good, just making that argument.

superjax, 2017-08-01:
Thanks@dpkoch.  I agree, it is nice to have a good examples, and an isolated case for launching the mav in sim is probably important, especially since gazebo launch files are so weird. I think that base.launch actually does that already, and that multirotor.launch and fixedwing.launch are pretty superfluous. (They each configure a single line of base.launch).

I'm wondering if it wouldn't make sense to throw rosflight_io into fixedwing and multirotor launch.  I'm not as inclined to include RC, because that's use-case specific and not a part of rosflight_pkgs, but rosflight_io is, so it makes sense to at least have an example of how that works.

My personal belief is that launch files packaged with nodes are primarily meant as examples.  That's at least how I've always used them, and how they are documented.  In other packages.  They start up a basic configuration of nodes and are a starting point for a user to customize to meet their needs.  It's not entirely obvious to a new user how the UDP connection works, which is why I think it ought to be included as an example in a launch file.

bsutherland333, 2023-10-04:
Launch files have been rewritten and are (hopefully) less confusing.

closed 2023-10-04 by bsutherland333