rosflight/rosflight_ros_pkgs #41: Airspeed sensor setup

sethholsinger, 2017-08-09:
How did you guys setup your airspeed sensor with the naze32? I haven't found any good information on how to do this anywhere. I am using the same airspeed sensor that you recommended here: [http://docs.rosflight.org/en/latest/user-guide/hardware-setup/](http://docs.rosflight.org/en/latest/user-guide/hardware-setup/).

gellings, 2017-08-09:
It is just an I2C connection (including scl, sda, 5V, gnd).  Your naze should break out the correct pins. They are probably some un-soldered pads or through holes. I recommend doing a quick internet search for the pinout of your board version. You just need to solder some kind of connection with the airspeed sensor wire. I think we just used some male headers and put a female header on the wire, but more reliable connectors are obviously preferable.

sethholsinger, 2017-08-09:
Thanks! That is what I thought but I wasn't sure and couldn't find a definitive answer.

closed 2017-08-09 by sethholsinger