rosflight/rosflight_ros_pkgs #42: Airspeed is negative and differential pressure is infinite

sethholsinger, 2017-08-10:
I hooked up an airspeed sensor and the readings that I am getting for airspeed and differential pressure are starting extremely high (differential pressure was x10^19) close to start up of the naze32. These readings then increase to until differential pressure becomes infinite at which point velocity becomes a very small negative number (ros reads -0.0). This is seems to be a problem with the firmware where the readings are being added together over and over but I have not confirmed that yet. Temperature readings seem fine though.

superjax, 2017-08-10:
@drinkdhmo, @dallinbriggs,  Did you observe this behavior on your last flight?

drinkdhmo, 2017-08-10:
No, but we flew the 'drinkdhmo:sensor_outlier_rejection' branch of firmware on Friday.

superjax, 2017-08-10:
Hmm..  let's merge they in soon.

superjax, 2017-08-10:
Okay, so I just flashed a clean build of the `rc1.0` branch.  Here is my output

[image]

The noise on the sensor pretty large around 0.  (Like +/- 8 m/s)  It seems to be a lot less noisy when real airspeed is being measured.  There is absolutely no filtering on the airspeed measurement, so you'll want to do some pretty heavy low-pass filtering in your code.

@drinkdhmo, I think that your `sensor_outlier_rejection` branch is using the latest breezy.  Is that correct?

drinkdhmo, 2017-08-10:
That is correct. Just waiting on Dan to merge.

superjax, 2017-08-11:
Okay.  So, @sethholsinger, have you tried the latest `rc1.0` branch?  It seems to be working pretty well for me.  

I remember seeing issues like you're seeing.  They tend to be a problem with old versions of Breezy - the F1 driver layer.  Perhaps try a `git submodule update` after pulling the latest `rc1.0` branch.  It could be that you are flashing an old version of the board drivers.

sethholsinger, 2017-08-11:
Thank you all for your input. The good new is that flashing the latest version of the firmware allows me to see both temperature and differential pressure readings. Unfortunately I don't get any velocity readings. ROS just always outputs 0,0.

superjax, 2017-08-11:
That's weird....  

Which topic are you reading no velocity on?

sethholsinger, 2017-08-11:
/airspeed

superjax, 2017-08-11:
hmmm...  Which version of ubuntu are you on?

sethholsinger, 2017-08-11:
I downloaded a fresh version from master and that is when I got the errors that I mention in my last post. I used 14.04.

superjax, 2017-08-11:
Could you please post the output of `catkin_make`?  14.04 ships with a different version of eigen than 16.04.  It's super annoying, and can cause weird compilation issues.

Also, try commenting out lines 87 - 97 of `rosflight/rosflight/CMakeLists.txt`

(https://github.com/rosflight/rosflight/blob/master/rosflight/CMakeLists.txt#L87)[https://github.com/rosflight/rosflight/blob/master/rosflight/CMakeLists.txt#L87]

sethholsinger, 2017-08-14:
The error is with eigen3 though I do have eigen3. I would post the error but it is extremely long. So here is a some of the error:
/usr/include/eigen3/Eigen/src/Core/ProductBase.h:91:34:   required from ‘Eigen::ProductBase<Derived, Lhs, Rhs>::ProductBase(const Lhs&, const Rhs&) [with Derived = Eigen::GeneralProduct<Eigen::GeneralProduct<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > >, Eigen::Transpose<Eigen::Matrix<double, -1, -1> >, 5>, Eigen::Matrix<double, -1, -1>, 5>; Lhs = Eigen::GeneralProduct<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > >, Eigen::Transpose<Eigen::Matrix<double, -1, -1> >, 5>; Rhs = Eigen::Matrix<double, -1, -1>]’
/usr/include/eigen3/Eigen/src/Core/products/GeneralMatrixMatrix.h:391:66:   required from ‘Eigen::GeneralProduct<Lhs, Rhs, 5>::GeneralProduct(const Lhs&, const Rhs&) [with Lhs = Eigen::GeneralProduct<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > >, Eigen::Transpose<Eigen::Matrix<double, -1, -1> >, 5>; Rhs = Eigen::Matrix<double, -1, -1>]’
/usr/include/eigen3/Eigen/src/Core/GeneralProduct.h:595:91:   required from ‘const typename Eigen::ProductReturnType<Derived, OtherDerived>::Type Eigen::MatrixBase<Derived>::operator*(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::Matrix<double, -1, -1>; Derived = Eigen::GeneralProduct<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > >, Eigen::Transpose<Eigen::Matrix<double, -1, -1> >, 5>; typename Eigen::ProductReturnType<Derived, OtherDerived>::Type = Eigen::GeneralProduct<Eigen::GeneralProduct<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > >, Eigen::Transpose<Eigen::Matrix<double, -1, -1> >, 5>, Eigen::Matrix<double, -1, -1>, 5>]’
/home/seth/flight_ws/src/rosflight/rosflight/src/mag_cal.cpp:454:59:   required from here
/usr/include/eigen3/Eigen/src/Core/util/BlasUtil.h:213:60: error: no matching function for call to ‘Eigen::internal::blas_traits<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > > > >::extractScalarFactor(const type&)’
   { return - Base::extractScalarFactor(x.nestedExpression()); }
                                                            ^
/usr/include/eigen3/Eigen/src/Core/util/BlasUtil.h:213:60: note: candidate is:
/usr/include/eigen3/Eigen/src/Core/util/BlasUtil.h:170:30: note: static const Scalar Eigen::internal::blas_traits<XprType>::extractScalarFactor(const XprType&) [with XprType = Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > >; Eigen::internal::blas_traits<XprType>::Scalar = double]
   static inline const Scalar extractScalarFactor(const XprType&) { return Scalar(1); }
                              ^
/usr/include/eigen3/Eigen/src/Core/util/BlasUtil.h:170:30: note:   no known conversion for argument 1 from ‘const type {aka const Eigen::Matrix<double, -1, -1>}’ to ‘const Eigen::ReturnByValue<Eigen::internal::inverse_impl<Eigen::Matrix<double, -1, -1> > >&’
make[2]: *** [rosflight/rosflight/CMakeFiles/calibrate_mag.dir/src/mag_cal.cpp.o] Error 1
make[1]: *** [rosflight/rosflight/CMakeFiles/calibrate_mag.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
Removing those lines of code worked and it built. Thanks!

sethholsinger, 2017-08-14:
Okay I tried getting new version of rosflight and new firmware working together and the imu is unresponsive on the new version of the firmware (I reflashed the old version to make sure the imu is still good). I am getting differential pressure for the /airspeed topic, but no velocity on the new firmware.

superjax, 2017-08-14:
@dpkoch, can you look into this error?  I don't have a v5 naze with me, but there is one in my desk.

sethholsinger, 2017-08-14:
Just a bit of additional information I tried this on both 14.04 (on a virtual machine) and 16.04 (on an odroid xu4). In order to get both of them to work I had to delete the rosflight_sim package and comment out the lines  87 - 97 of rosflight/rosflight/CMakeLists.txt as suggested by @superjax.

superjax, 2017-08-15:
Quick question.  When you say the IMU is unresponsive, do you mean you are getting the "IMU not responding error?". 

Could you check to see if you are actually receiving IMU messages?  I just noticed a bug where that error gets thrown and immediately cleared when reading or writing parameters, because the parameter write and read takes too long.

superjax, 2017-08-15:
Sorry about the eigen compilation issue.  Perhaps we should just add into the cmakelists a check on the eigen version and disable that code if it's going to break.

sethholsinger, 2017-08-17:
Yes, I am receiving imu messages but its just displays all values as zero over and over again or the last message sent to ros master of that type. This also is happening with /rc_raw,

superjax, 2017-08-17:
That's fine.  I'm actually out of town, so I don't have any way to test the v5 until Monday anyway.

This issue with the accel has me stumped.   I won't have time to look at it until Monday, but @len0rd, @cmcquinn or @dpkoch may have time to look at it before me.  Could you post 

1. Your parameters.yaml
2. Hardware config (board/sensor configuration)
3. Compiler version (`gcc-arm-none-eabi --version`)

Let's see if we can reproduce the issue and help you out.

superjax, 2017-08-17:
I'm sorry, I've been working from my phone, and accidentally closing this issue.  Haha.

sethholsinger, 2017-08-18:
Parameters are the stock parameters I didn't change them at all after the flash. I have tried both rev 5 and 6 naze32 but I have only hooked up an airspeed sensor to the rev 6.Both of these boards have barometers on them. My compiler version for g++/gcc is 4.8.4.

bsutherland333, 2023-10-04:
Issue is not observed on currently supported hardware.

closed 2023-10-04 by bsutherland333