rosflight/rosflight_ros_pkgs #55: Incomplete Dependencies?

mmmfarrell, 2017-09-14:
After following the ROS setup instructions in the docs, more dependencies were still necessary. 

I am running Ubuntu 16.04.2 LTS with Ros Kinetic on an Odroid XU4.

When trying to catkin_make a workspace that only contains rosflight, the error is this:

  ```
Could not find a package configuration file provided by "eigen_stl_containers"
  with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake
```
This error was solved by installing the ros kinetic eigen-stl-containers package with:

`sudo apt-get install eigen-stl-containers`

closed 2019-07-22 by superjax

Arsalan66, 2019-11-17:
sir,even after running the command,i'm still getting the error in ros melodic

dpkoch, 2019-11-19:
You need to do `sudo apt install ros-melodic-eigen-stl-containers`

Arsalan66, 2019-11-19:
Thanks for the kind response it worked however i am getting these errors 
[image]
and my simulation is empty, kindly help me resolve this
[image]

superjax, 2019-11-20:
These are C++ compiler errors from `ibvs_sim`.  I don't know what this is, and is unlikely related to `rosflight`.

If I were you, I would very carefully read each compiler error (the compiler is telling you what is wrong) and solve them one by one.  `stackoverflow.com` is a great resource for solving compiler errors.

This may seem like an insurmountable task at first, but we have all been there, and it's really the only way to learn.  Good luck!