rosflight/rosflight_ros_pkgs #59: Topic /baro displays a constant altitude, press, temperature data

biswasse, 2017-10-06:
Hi everyone! I'm new to rosflight, I am using the AfroFlight Naze32 Flight Controller (link: https://hobbyking.com/en_us/naze32r6-set-pcb-wire-pins.html) which has the BMP280 barometer onboard. OS version - Ubuntu 16.04; git tag - master, parameter file - rosservice call /param_write, hardware configuration - fixed wing.

The issue is that when I echo the /baro topic after running 'rosrun rosflight rosflight_io', I get a constant altitude, pressure and temperature even when I change the altitude of the controller. The imu data topic seems to be echoing the data correctly, however, I'm unable to understand why /baro returns fixed data for all the message types.

Let me know how I can solve this issue.

Thanks, 
Srijanee
[image]

superjax, 2017-10-07:
Hey @biswasse!  Welcome!

We don't have any naze32's with the BMP280, so I haven't written a driver for it.  The unusual thing is that the BMP280 is on the same I2C address as the MS5611, which we do support.  I would need to dig a little bit into the BMP280 datasheet to see how to support both barometers.  I'll get back to you on it, (this would be a modifcation to the BreezySTM32 hardware support layer)

superjax, 2017-10-07:
Just doing a little digging.  While the BMP280 and MS5611 _can_ have the same address, they have a pin that changes their address, depending on whether it's pulled down or up.  Cleanflight source code indicates that on naze32's with the BMP280, the address has been switched from `0x76` (which is the default address for the ms5611 on rev 5 and earlier naze32s) to 0x77. 

Is there any chance you would be able to run the `i2csniff` example in breezy?  We have a couple of simple hardware unit test type of things in breezy, all you have to do is the following steps:

```bash
git clone https://github.com/rosflight/BreezySTM32.git
cd BreezySTM32/examples/i2csniff
# <-- plug in the naze, with the boot pins shorted -->
make flash
miniterm.py /dev/ttyUSB* 115200 # <--- put the right USB device here
```

You should see something like

``` bash
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Found device at address 0X1E
Found device at address 0X68
Found device at address 0X77
--------------------------
```

You don't have to use miniterm, you could use screen or something like that, but the `i2cniff` example just tries all the i2c addresses and reports which ones get a `NACK`.  If you could do that, then it should be pretty easy to port the cleanflight bmp280 driver to our asynchronous method.

biswasse, 2017-10-12:
Sorry for the delayed response. I was out of town. I tried to do what you suggested but 'make flash' throws up an error. I have attached a screen shot of the same for your reference. Let me know how I can resolve this issue.

I also git cloned the pySerial repository which has the miniterm.py code, I am not sure whether I need this. Also, I realized that all the newer naze32's have the BMP280 sensor on-board. Is it possible to provide driver support for the same ?

Regards,
Srijanee Biswas
[image]

superjax, 2017-10-12:
Oh, sorry about that, it looks like you don't have the `stm32flash` CLI utiltiy installed.

We have 2 options:

You could follow this tutorial to install the build tools:

http://docs.rosflight.org/en/latest/developer-guide/building-flashing/

or use the cleanflight gui to flash this `.hex` file.
[myproject.hex.tar.gz](https://github.com/rosflight/rosflight/files/1380659/myproject.hex.tar.gz)

superjax, 2017-10-12:
I think I found a bmp280 lying around, so if I can just get the i2c address nailed down, I'm pretty sure we can get the driver working no problem.

You're right, it looks like a lot of the newer boards are using it, so it seems like a good idea to support it.

biswasse, 2017-10-12:
Sorry for bothering you with this again but I am new to this hardware system. I installed the stm32 flash utility and then tried to run 'make flash' command from my i2csniff directory, it gave me another error. I have attached  a screenshot for your reference.

Another question, if I use the Cleanflight gui, which firmware version should I choose for NAZE?

Thanks,

Srijanee Biswas

http://docs.rosflight.org/en/latest/developer-guide/building-flashing/

superjax, 2017-10-12:
No worries!  Thanks for pointing out that we are missing the driver!

I think I am missing the screenshot.  (maybe an issue with replying from email?) 

As for the firmware version, it shouldn't matter.  You should choose `'Load Firmware [Local]'`  The firmware version is talking about which version of the cleanflight firmware it is going to download to flash.

biswasse, 2017-10-12:
I have attached it again. Let me know whether you can view it. I can update the same on the GitHub issue as well.

Thanks,

Srijanee

superjax, 2017-10-12:
Yeah, it's still missing.  Perhaps you need to upload it directly to the issue.

biswasse, 2017-10-12:
[image]

Sorry about that. Please find attached.

superjax, 2017-10-12:
no problem.

Ah, the `failed to init device` error.  That typically happens for one of the following reasons:
 1. the naze is not in bootloader mode
 2. if you try to flash too quickly after plugging in the naze (even if it is in bootloader mode, it can take a second or two to respond)
 3. you are trying to talk to the wrong device, or you don't have permissions to flash

One way to tell that you're in bootloader mode is that only the power LED will be on, and no other LEDs will flash.

Can you verify that none of those things are the issue?

superjax, 2017-10-12:
by the way, you can flash to a device other than `/dev/ttyUSB0` by supplying the `SERIAL_DEVICE` variable

`make flash SERIAL_DEVICE=/dev/ttyUSB1`

biswasse, 2017-10-12:
I have both the power LED on as well the red LED flashing when I plug in the device to my computer's USB port. I'm not trying to flash too quickly after plugging it in. So, does it mean I'm facing the third issue? How do I verify the device address and whether or not I have the permission to flash?

Thanks,

Srijanee

superjax, 2017-10-12:
Yeah, it looks like you're not shorting the boot pins when you power up the naze.    It can be kinda tough, but you have to short these pins when you plug in the USB.

[image]

biswasse, 2017-10-12:
I seem to have flashed the firmware successfully using the Cleanflight configurator (the activity LED not flashing when I powered up the board with the boot pins shorted). The progress bar showed a successful completion. Now, when I power off the board, remove the short and power it up again, the red LED doesn't flash anymore.

What should I do next ?

[picture]<https://camo.githubusercontent.com/4e500ab0cd85281cfc5cfda535e6a07763f41226/687474703a2f2f7777772e64726f6e6574726573742e636f6d2f75706c6f6164732f6462353239302f6f7074696d697a65642f32582f392f393737613261333932346366623030346363643934323232386539366534363537613239326235625f315f363839783338352e6a7067>

biswasse, 2017-10-12:
Additionally, when I now run the command - 'rosrun rosflight rosflight_io', it gives the following message -

[ INFO] [1507848422.806133979]: Received 0 of 0 parameters. Requesting missing parameters...

/imu/data and /baro also don't seem to be getting published now which was not the case before when I opened this issue (that's how I saw /baro topic was echoing constant values).

Let me know if I have made any errors.

Thanks

Srijanee

________________________________
From: Biswas, Srijanee (biswasse)
Sent: Thursday, October 12, 2017 3:45:33 PM
To: rosflight/rosflight; rosflight/rosflight
Cc: Mention
Subject: Re: [rosflight/rosflight] Topic /baro displays a constant altitude, press, temperature data (#59)

I seem to have flashed the firmware successfully using the Cleanflight configurator (the activity LED not flashing when I powered up the board with the boot pins shorted). The progress bar showed a successful completion. Now, when I power off the board, remove the short and power it up again, the red LED doesn't flash anymore.

What should I do next ?

superjax, 2017-10-12:
Yeah, now you have the example flashed to the board, try running `miniterm.py` or `screen` or something to see what the naze is printing out.

superjax, 2017-10-12:
the exact command I use for running `miniterm` is 

``` bash
miniterm.py /dev/ttyUSB0 115200
```

biswasse, 2017-10-12:
When I run the above command, it prints the following

Found device at address 0X68
Found device at address 0X76

biswasse, 2017-10-12:
Should I git clone the rosflight package once again?

superjax, 2017-10-12:
You got it!  haha!  Thanks!

So, as you just now discovered, it looks like the `bmp280` is on device address `0x76`, and not on `0x77`.  That's a good thing, because it means we can tell a difference between the `ms5611` and the `bmp280`.  

Going forward - myself or one of the guys on the team (or you can if you want) will have to modify BreezySTM32 to detect and use the `bmp280`.  That might take a few days, unfortunately, since we are all graduate students and don't have a ton of time.  I'm going to change the title of this issue to reflect the actual problem, and hopefully we can get a solution going pretty quick.

Thanks for the help!

superjax, 2017-10-12:
By the way, if you want to use `rosflight_io` again, you'll have to flash the `rosflight.hex` file again.  The example hex I had you flash is a super simple piece of code to sniff the i2c port, it's not the rosflight firmware.  That's why `rosflight_io` reports no parameters.

biswasse, 2017-10-12:
Oh, I get it. My bad. I'll flash the rosflight.hex file again and see if I can echo all the topics correctly.

Thanks a lot for your help.

Regards,
Srijanee Biswas

biswasse, 2017-10-13:
Hi James,

I know you said that it might take a few days to modify BreezySTM32 to detect and use the bmp280, but I was wondering how long that might be.

We actually bought a couple of naze32s for a graduate class project this semester, so, it'll be great if you can suggest a work-around (maybe, flip32s will have the barometer support?) so that we can utilize the same.

Thanks and regards,

Srijanee

superjax, 2017-10-13:
The flip32s do have barometer support.  I don't that I'll have time to fix it till Tuesday at the earliest.

superjax, 2017-10-14:
Hey, bad news.  Turns out my bmp280 board doesn't work...  I'll have to order a new one.  I can try and develop the driver if you don't mind flashing the binaries and testing it for me.  But the earliest I'm going to get a new board is Wednesday.

biswasse, 2017-10-14:
Sure, I can do that if you're willing to guide me through it.

Srijanee Biswas

superjax, 2017-10-17:
@sethholsinger, just wanted to bring you in on this discussion.

superjax, 2017-10-17:
So, @biswasse.  I have a proof-of-concept example for the bmp280.  Could you try this example on your rev6?

[myproject.hex.tar.gz](https://github.com/rosflight/rosflight/files/1392892/myproject.hex.tar.gz)

This should just print out the raw pressure and temperature readings.

biswasse, 2017-10-17:
Hi James,

Unfortunately, I have already left my lab for the day. I'll be sure to try it tomorrow morning and let you know.

Thanks
Srijanee Biswas

superjax, 2017-10-17:
Awesome.  My board shows up tomorrow afternoon as well, so I should be able to test it myself tomorrow night and have a working solution by Thursday.

biswasse, 2017-10-18:
Hi James,

So, I tried flashing the new hex file on my rev6 board. Although, the Cleanflight gui shows a flash successful message but the activity led doesn't glow, neither does the rosflight_io node seem to be publishing anything.

Find a screenshot of the same.

[image]

Thanks,
Srijanee Biswas

superjax, 2017-10-18:
darn.  I probably did something wrong.  I'll work on it some more.

superjax, 2017-10-19:
Okay guys.  I don't have a rev6 naze (it's in the mail, but not going to get here in a while). I tried this with my bmp280 breakout board with my rev2.  Would one you mind flashing this?

[rosflight_bmp280.hex.tar.gz](https://github.com/rosflight/rosflight/files/1397091/rosflight_bmp280.hex.tar.gz)

sethholsinger, 2017-10-19:
So it works but the altitude readings drift in the positive direction significantly. In about a minute I have seen the value drift by as much as 16 meters.

superjax, 2017-10-19:
Does it level off?  I'm unfamiliar with this sensor a little bit, and sometimes there is a bit of a temperature bias that settles out after a few minutes of being on.

sethholsinger, 2017-10-19:
Doesn't seem to I have it running in the background now it keeps increase but it does seem to have slowed down a bit. I will wait a bit and see if is settles at some point.

superjax, 2017-10-19:
Glad it's working though!

superjax, 2017-10-19:
K let me know if it doesn't.  I don't really know what to do off the top of my head to fix a drift like you're talking about, but we can give it a try.  Would you mind also looking at the pressure and temperature?  Altitude is approximated using those two measurements, the baro doesn't actually measure altitude.

sethholsinger, 2017-10-19:
Temperature does seem to change much. Pressure seems to be drifting in the negative direction.

sethholsinger, 2017-10-19:
Hmm actual as the temperature goes up the pressure seems to be dropping.

sethholsinger, 2017-10-19:
It has slowed significantly. I am going to recalibrate the barometer wait 10 minutes then tell you what value it at.

sethholsinger, 2017-10-19:
It drifted less than a meter. It seems like it just needed to warm up. Thanks for getting this sensor working!

superjax, 2017-10-19:
Cool.  Thanks for testing it out for me!

superjax, 2017-10-19:
By the way, the warm-up period is typical of all MEMS sensors.  I've seen it pretty bad with the IMU and mag as well, and it can mess up things if you're not careful.  You can calibrate a linear thermal compensation constant which helps a lot.  We used to do that, but it was pretty complicated.  Just something to be aware of.

superjax, 2017-10-19:
@biwasse, can you confirm that this fix worked for you?

biswasse, 2017-10-20:
Hi James,

Thanks so much for the help. Yes it works for me as well.

closed 2018-01-10 by superjax