rosflight/rosflight_ros_pkgs #60: GPS Node Hard Coded Port Issue

sethholsinger, 2017-10-18:
I found that in your GPS code has hard coded the port to access the GPS with as /dev/ttyUSB0. On the Odroid XU4 running Ubuntu 16.04 Mate I was using to test the GPS on this causes an issue as the port the GPS is connected on is port /dev/ttyACM0. It is a pretty easy fix on my end as I just changed your code to access /dev/ttyACM0 instead of /dev/ttyUSB0.

superjax, 2017-10-19:
I believe you should be able to change it with a ROS parameter.  Here are the lines from the `gps` node

``` python
    serial_port = rospy.get_param('~port','/dev/ttyUSB0')
    serial_baud = rospy.get_param('~baud',115200)
```

I think this means that you should be able to specify the `port` parameter

`rosrun rosflight_utils gps _port:=/dev/ttyACM0`

I can verify this tomorrow, but does this work for you?

sethholsinger, 2017-10-19:
Sorry, I wasn't aware that could be done. I have just changed the value to /dev/ttyACM0 though so I don't have to type that in every time.

closed 2017-10-19 by sethholsinger