rosflight/rosflight_ros_pkgs #75: Can't get multirotor to fly in Gazebo

msadowski, 2018-04-20:
Hi!

I was just trying to follow the docs to run Gazebo simulation with rosflight I didn't manage to make it work. 

Some information:
1. OS Version: Ubuntu 16.04.4
2. Git tag: master

3. Workspace structure
I clone the rosflight and rosflight_joy repos into src directory of my workspace. When running catkin_make everything compiles cleanly and after sourcing the devel I can run the nodes.

4.Joystick: FrSky Taranis

First thing that is different from the steps in the docs is that rqt_graph doesn't launch automatically. The rqt structure I see after running all the nodes and launch files is this:
[image]

Also in Gazebo the undercarriage of the drone is below the floor level (seems it doesn't have collisions enabled?)

Even though I got everything to run I see error messages when trying to arm the quad. I made a call to calibrate_imu and calibrate_baro and the error in the rosflight_io is 
```
[ERROR] [1524214501.767272762, 464.493000000]: [Autopilot]: unable to arm due to error code 0x1
```

I was trying to find some information about this error code in the docs and even source code but didn't find any useful information. Would you be able to point me in some direction?

Thanks!

dpkoch, 2018-04-20:
Error code 0x01 is "invalid mixer." We're working on improving those error messages and the docs so they're more useful, but for now the errors are defined in the firmware at [include/state_manager.h:67](https://github.com/rosflight/firmware/blob/master/include/state_manager.h#L67-L76). You can fix that by setting the appropriate mixer with `rosservice call /param_set MIXER 1`.

If you don't want to go through the calibration and setting the mixer every time you start the simulation, do those steps once and then call `rosservice call /param_write` just like you would with a hardware autopilot. That command saves what would be the flash memory of the autopilot to a file (`~/.ros/rosflight_memory/multirotor/mem.bin`) and then subsequent runs of the simulator will load that on startup.

As far as collisions, it looks like they're enable for the actual body of the multirotor, but not the landing gear. So I think you should be good there. `rqt_graph` not launching automatically should also be okay, that's actually normal behavior.

superjax, 2019-07-22:
Closing due to inactivity

closed 2019-07-22 by superjax