rosflight/rosflight_ros_pkgs #82: FOUND Sonar

mmmfarrell, 2018-11-09:
Nvidia TX2 with Oribitty Carrier Board Running Ubuntu 16.04

Running the master branch in the motion capture room.

Autopilot said it FOUND SONAR and I don't have a sonar or any external devices plugged in.

Just leaving this here because that shouldn't happen haha

mmmfarrell, 2018-11-09:
Also when I launch rosflight_io I get Autopilot now it Rate mode. But its actually in angle mode. RC_ATT_CTRL_CHN = -1 and RC_ATT_MODE = 1. And I actually fly it and it is definitely in angle mode.

bsutherland333, 2023-10-16:
Our current firmware hasn't had this issue, and we won't be doing any more work on previous board implementations so I'm going to close this.

closed 2023-10-16 by bsutherland333