rosflight/rosflight_ros_pkgs #88: Error : Unhealthy estimator

TomRvr, 2019-01-02:
Hello !
I try to fly a multi-rotor with gazebo simulation with the following configuration :
OS Version : Ubuntu 16.04
ROS Kinetic
Git tag : master
Xbox controller

When I want to fly, all the nodes seems to be started correctly.
**rqt_graph**
[image]

On gazebo, I can arm the drone but when I want to slowly increase the throttle, the drone is uncontrollable and the Unhealthy estimator error occur :
**Terminal**
[image]

I might forget to configure one parameter but I can't find which one.
**Parameter file**
```
{name: ACC_LPF_ALPHA, type: 9, value: 0.8999999761581421}
{name: ACC_X_BIAS, type: 9, value: -1.542290462008822e-10}
{name: ACC_X_TEMP_COMP, type: 9, value: 0}
{name: ACC_Y_BIAS, type: 9, value: -1.789761672199575e-10}
{name: ACC_Y_TEMP_COMP, type: 9, value: 0}
{name: ACC_Z_BIAS, type: 9, value: 0.006649971008300781}
{name: ACC_Z_TEMP_COMP, type: 9, value: 0}
{name: AIL_REV, type: 6, value: 0}
{name: ARM_CHANNEL, type: 6, value: 5}
{name: ARM_SPIN_MOTORS, type: 6, value: 1}
{name: ARM_THRESHOLD, type: 9, value: 0.1500000059604645}
{name: BARO_BIAS, type: 9, value: 1.005905389785767}
{name: BAUD_RATE, type: 6, value: 921600}
{name: CAL_GYRO_ARM, type: 6, value: 1}
{name: DIFF_PRESS_BIAS, type: 9, value: 0}
{name: ELEVATOR_REV, type: 6, value: 0}
{name: FAILSAFE_THR, type: 9, value: 0.300000011920929}
{name: FILTER_INIT_T, type: 6, value: 3000}
{name: FILTER_KI, type: 9, value: 0.009999999776482582}
{name: FILTER_KP, type: 9, value: 0.5}
{name: FILTER_MAT_EXP, type: 6, value: 1}
{name: FILTER_QUAD_INT, type: 6, value: 1}
{name: FILTER_USE_ACC, type: 6, value: 1}
{name: FIXED_WING, type: 6, value: 0}
{name: GROUND_LEVEL, type: 9, value: 1387}
{name: GYRO_LPF_ALPHA, type: 9, value: 0.300000011920929}
{name: GYRO_X_BIAS, type: 9, value: -1.826129802928733e-10}
{name: GYRO_Y_BIAS, type: 9, value: 1.573581126512735e-10}
{name: GYRO_Z_BIAS, type: 9, value: 1.141862218800191e-19}
{name: MAG_A11_COMP, type: 9, value: 1}
{name: MAG_A12_COMP, type: 9, value: 0}
{name: MAG_A13_COMP, type: 9, value: 0}
{name: MAG_A21_COMP, type: 9, value: 0}
{name: MAG_A22_COMP, type: 9, value: 1}
{name: MAG_A23_COMP, type: 9, value: 0}
{name: MAG_A31_COMP, type: 9, value: 0}
{name: MAG_A32_COMP, type: 9, value: 0}
{name: MAG_A33_COMP, type: 9, value: 1}
{name: MAG_X_BIAS, type: 9, value: 0}
{name: MAG_Y_BIAS, type: 9, value: 0}
{name: MAG_Z_BIAS, type: 9, value: 0}
{name: MIN_THROTTLE, type: 6, value: 1}
{name: MIXER, type: 6, value: 2}
{name: MOTOR_IDLE_THR, type: 9, value: 0.1000000014901161}
{name: MOTOR_MAX_PWM, type: 6, value: 2000}
{name: MOTOR_MIN_PWM, type: 6, value: 1000}
{name: MOTOR_PWM_UPDATE, type: 6, value: 490}
{name: OVRD_LAG_TIME, type: 6, value: 1000}
{name: PARAM_MAX_CMD, type: 9, value: 1}
{name: PID_PITCH_ANG_D, type: 9, value: 0.05000000074505806}
{name: PID_PITCH_ANG_I, type: 9, value: 0}
{name: PID_PITCH_ANG_P, type: 9, value: 0.1500000059604645}
{name: PID_PITCH_RATE_D, type: 9, value: 0}
{name: PID_PITCH_RATE_I, type: 9, value: 0}
{name: PID_PITCH_RATE_P, type: 9, value: 0.07000000029802322}
{name: PID_ROLL_ANG_D, type: 9, value: 0.05000000074505806}
{name: PID_ROLL_ANG_I, type: 9, value: 0}
{name: PID_ROLL_ANG_P, type: 9, value: 0.1500000059604645}
{name: PID_ROLL_RATE_D, type: 9, value: 0}
{name: PID_ROLL_RATE_I, type: 9, value: 0}
{name: PID_ROLL_RATE_P, type: 9, value: 0.07000000029802322}
{name: PID_TAU, type: 9, value: 0.05000000074505806}
{name: PID_YAW_RATE_D, type: 9, value: 0}
{name: PID_YAW_RATE_I, type: 9, value: 0}
{name: PID_YAW_RATE_P, type: 9, value: 0.25}
{name: RC_ATT_CTRL_CHN, type: 6, value: 4}
{name: RC_ATT_MODE, type: 6, value: 1}
{name: RC_ATT_OVRD_CHN, type: 6, value: 6}
{name: RC_F_CHN, type: 6, value: 2}
{name: RC_MAX_PITCH, type: 9, value: 0.7860000133514404}
{name: RC_MAX_PITCHRATE, type: 9, value: 3.141590118408203}
{name: RC_MAX_ROLL, type: 9, value: 0.7860000133514404}
{name: RC_MAX_ROLLRATE, type: 9, value: 3.141590118408203}
{name: RC_MAX_YAWRATE, type: 9, value: 1.506999969482422}
{name: RC_NUM_CHN, type: 6, value: 6}
{name: RC_OVRD_DEV, type: 9, value: 0.1000000014901161}
{name: RC_THR_OVRD_CHN, type: 6, value: 6}
{name: RC_TYPE, type: 6, value: 1}
{name: RC_X_CHN, type: 6, value: 0}
{name: RC_Y_CHN, type: 6, value: 1}
{name: RC_Z_CHN, type: 6, value: 3}
{name: RUDDER_REV, type: 6, value: 0}
{name: STRM_AIRSPEED, type: 6, value: 20}
{name: STRM_ATTITUDE, type: 6, value: 200}
{name: STRM_BARO, type: 6, value: 50}
{name: STRM_HRTBT, type: 6, value: 1}
{name: STRM_IMU, type: 6, value: 500}
{name: STRM_MAG, type: 6, value: 50}
{name: STRM_RC, type: 6, value: 50}
{name: STRM_SERVO, type: 6, value: 50}
{name: STRM_SONAR, type: 6, value: 40}
{name: STRM_STATUS, type: 6, value: 10}
{name: SWITCH_5_DIR, type: 6, value: 1}
{name: SWITCH_6_DIR, type: 6, value: 1}
{name: SWITCH_7_DIR, type: 6, value: 1}
{name: SWITCH_8_DIR, type: 6, value: 1}
{name: SYS_ID, type: 6, value: 1}
{name: X_EQ_TORQUE, type: 9, value: 0}
{name: Y_EQ_TORQUE, type: 9, value: 0}
{name: Z_EQ_TORQUE, type: 9, value: 0}
```
I see this error is on the forum ( https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwif7oCfw8_fAhUB0RoKHaVSDokQFjAAegQIABAB&url=https%3A%2F%2Fdiscuss.rosflight.org%2Fdiscussion%2F11%2Funhealthy-estimator&usg=AOvVaw0LRiXSyyI0ZVY3OH5SQmpW ) but the forum seems to be down for the moment and I can't access to the topic.

I search for some help to indicate me the things to checks to find a solution.
Thanks !

TomRvr, 2019-01-16:
Thank you for your very clear answer.
Now I understand why this error happened.

Congrats for your work on this project.

superjax, 2019-01-16:
No Problem!  Thanks!

closed 2019-01-16 by superjax