rosflight/rosflight_ros_pkgs #98: Sent pwm from onboard computer

JonathanPlasse, 2019-07-10:
Hi,
Is it possible to sent pwm signal to the motor every sample time from the onboard computer?
Is MAVLINK a possibility?
Thanks in advance

superjax, 2019-07-10:
I haven't taken a close look at the USB latency between the onboard computer and the flight controller.  It may be possible, but it might also be too laggy, depending on the dynamics of your platform.

The fixed-wing mixers respond at 50Hz, and we fly planes straight from the onboard computer by passing motor commands directly.  That seems to work just fine.

Multirotors, on the other hand, have much faster dynamics.  Their motors update at 490 Hz, so while is certainly enough bandwidth to support it, if the onboard computer is not running an RTOS, you can end up with inconsistent real-time response at the microsecond resolution.

Quick bandwidth calculation: 490Hz * 14 channels * 16 bits =~ 110 kb/s, and the USB peripheral supports up to at least 3000000 baud, so there is plenty of bandwidth, as long as the packets are sent at regular intervals.

superjax, 2019-07-10:
Also, if you're interested in pursuing this further, @willisj2 is working on merging in the "aux command" functionality into the firmware.  This will open up access to all 14 outputs.

willisj2, 2019-07-10:
Yeah, the /aux_command topic should do what you want - though you won't have much control over update rates. It would work well for servos, but you'll probably find some jitter with ESCs. It can be used in conjunction with one of the mixers, so, for example, if you want to fly a hexacopter with landing gear or pwm led strips, you could use the Hex-Plus mixer and write to channels 7 and 8 using the aux_command.

JonathanPlasse, 2019-07-12:
Thanks for your answers.
Apparently, the latency of the usb serial bus could be reduced using either
`setserial /dev/<usb_device> low_latency`
or
`sudo bash -c “echo 1 > /sys/bus/usb-serial/devices/<usb_device>/latency_timer"`

superjax, 2019-07-16:
Thanks!  That's a great find!

We should add this to the wiki.

JonathanPlasse, 2019-07-17:
Also, How do you use the `PASSTHROUGH` command?

JonathanPlasse, 2019-07-17:
Is it possible to use UART3 instead of usb?
As it is mentioned in the documentation that UART3 could be used.
It would avoid latency problem.

superjax, 2019-07-17:
It is.  UART3 is supported through the `SERIAL_DEVICE` parameter.  I think you need to set it to `3` https://github.com/rosflight/firmware/blob/master/boards/airbourne/airbourne_board.h#L91

This is definitely undocumented, untested functionality though, so use at your own risk!  ;)  It should work, though.  I've used it to fly remotely from a 3DR radio, but that was a couple of months ago and things may have changed.

Most x86 computers don't have a UART peripheral accessible in the kernel, so I've always had to use some kind of FTDI->USB converter, which introduces the same kind of latency.  I'm not sure if that exists on the TX2.

JonathanPlasse, 2019-07-19:
If this method is used to control the quadcopter, is it still possible to simulate a flight test in the SIL simulation?

dpkoch, 2019-07-19:
The passthrough command is documented [here](https://docs.rosflight.org/user-guide/autonomous-flight/#provide-control-from-a-companion-computer), is there anything missing that would be useful?

Whether you use USB or UART has no effect on the SIL simulation, because it simulates the serial connection with a UDP connection between the simulation "board" and the `rosflight_io` node

closed 2023-03-25 by JonathanPlasse