rosflight/rosflight_ros_pkgs #100: Can rosflight fly a swarm of drone?

JonathanPlasse, 2019-07-17:
Hi,
I have a project where I want to fly a swarm of drone.
Can ROSflight handle multiple drone?

superjax, 2019-07-17:
I'm not sure what you mean, but I can think of a few ways to accomplish this.  They are all going to be difficult, and may require modifying the firmware.

The first is to have an onboard computer for each MAV with some kind of WiFi connection.  Each onboard computer could have it's own `rosflight_io` running in a separate namespace.  I could see that working.

The second would be to use a 3DR radio for each MAV, and fly them all remotely through the 3DR connection with multiple copies of `rosflight_io` running on the base computer.

The one big thing I can see is that ROSflight does not support arming the MAV without an RC connection.  This is a safety feature, but it means that to fly a swarm, you'd need an RC transmitter for each MAV, which could get really expensive.  If _I_ were going to fly a swarm of MAVs, I'd probably just change the firmware so it didn't require an RC connection.  ;).

JonathanPlasse, 2019-07-18:
Thanks,
It is the first approach that I would like to take.
Another question,
Is it possible to simulate multiple vehicle with SIL in Gazebo?

superjax, 2019-07-19:
Totally,  you just need to ensure the UDP ports that each agent uses to communicate with the firmware running in SIL is unique and namespace everything correctly.

@jacobmoroni  has done this, you could ask him about it.

closed 2019-09-09 by dpkoch