rosflight/rosflight_ros_pkgs #113: Error installing on Raspbian Buster

rafaelmaeuer, 2019-10-20:
After a successful installation of ROS Melodic on a Raspberry Pi 3+ (Debian Buster) the installation of Rosflight fails when executing this command:
```
rosdep -y install --ignore-src --from-path rosflight
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rosflight_sim: No definition of [gazebo_ros] for OS version [buster]
rosflight_utils: No definition of [gazebo_msgs] for OS version [buster]
rosflight: No definition of [eigen_stl_containers] for OS version [buster]

Seems to me it's not compatible with the newest Raspbian (Buster) yet?

rafaelmaeuer, 2019-10-20:
OK it might be my fault, regarding this information from the docs:

> The only requirement for the companion computer is that it runs Linux (Ubuntu LTS versions 16.04 or 18.04)

So I need to run ubuntu on the Raspberry Pi?

superjax, 2019-10-21:
Um, you might be able to work around it.  You probably won't be able to install from repos with raspbian (as it might not have all the packages you need); however, you can probably just install the missing dependencies from source.

You actually don't _need_ gazebo.  Check out [rosflight_sim/CMakeLists](https://github.com/rosflight/rosflight/blob/master/rosflight_sim/CMakeLists.txt) lines 4 and 5.  If `gazebo` is not found, the whole simulation package is skipped silently.  We did this so that you can build `rosflight` from source on an SBC without needing the whole gazebo simulation stack.

You will need `eigen_stl_containers`, so just clone the [repo](https://github.com/ros/eigen_stl_containers) into your `catkin_ws/src` folder (just like any other ROS package) and it should build just fine.

rafaelmaeuer, 2019-10-21:
hey thank you so much, I installed the `eigen_stl_containers` as described, but the error message persists. So first, how can I exclude `gazebo` from my build, and secondly how can I tell it that `eigen_stl_containers` is already there?

superjax, 2019-10-21:
Don't use `rosdep` just try `catkin_make`

rafaelmaeuer, 2019-10-23:
When I run `catkin_make` I get following error: 
[image]

When I run `catkin_make_isolated` I get a permission denied error (running with sudo not possible)
```
IOError: [Errno 13] Permission denied: '/home/pi/ros_catkin_ws/build_isolated/.built_by'
```

superjax, 2019-10-23:
Based on the error message, it looks like there are non-catkin packages in your workspace.

what does your directory structure look like?  You should have
- catkin_ws
| - CMakeLists.txt 
| - src
| - | - rosflight
| - | - eigen_stl_containers

rafaelmaeuer, 2019-10-27:
in src folder there are way more packages (50?)
it looks like this:
```
- catkin_ws
| - src
| - | - actionlib
| - | - angles
...
| - | - CMakeLists.txt 
| - | - cmake_modules
...
| - | - eigen_stl_containers
...
| - | - rosflight
| - | - .rosinstall
...
| - | - webkit_dependency
| - | - xacro
```

rafaelmaeuer, 2019-11-09:
Never mind I succeeded installing ubuntu on the Pi so I can use the prebuild packages. But in general it would be interesting to use it with Raspbian too!

closed 2023-10-04 by bsutherland333