rosflight/rosflight_ros_pkgs #114: some errors in spawning quadcopter

Arsalan66, 2019-11-18:
1. OS Version: ubutu 18.04
2. Compiler version (amd 64)
sir, i ran catkin_make on the roscopter directory and it worked perfectly however when i ran catkin_make on ibvs sim it generated errors:
[ 96%] Building CXX object ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/plugins/src/boat_forces_and_moments.cpp.o
[ 97%] Built target ekf_rosbag
[ 98%] Built target ekf_node
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp: In destructor ‘virtual gazebo::BoatForcesAndMoments::~BoatForcesAndMoments()’:
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:29:18: error: ‘DisconnectWorldUpdateBegin’ is not a member of ‘gazebo::event::Events’
   event::Events::DisconnectWorldUpdateBegin(updateConnection_);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp: In member function ‘void gazebo::BoatForcesAndMoments::SendForces()’:
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:39:7: error: ‘math’ has not been declared
       math::Vector3(actual_forces_.Fx, actual_forces_.Fy, actual_forces_.Fz));
       ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:41:7: error: ‘math’ has not been declared
       math::Vector3(actual_forces_.l, actual_forces_.m, actual_forces_.n));
       ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp: In member function ‘void gazebo::BoatForcesAndMoments::UpdateForcesAndMoments()’:
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:166:3: error: ‘math’ has not been declared
   math::Pose W_pose_W_C = link_->GetWorldCoGPose();
   ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:167:15: error: ‘W_pose_W_C’ was not declared in this scope
   double pn = W_pose_W_C.pos.x;
               ^~~~~~~~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:170:3: error: ‘math’ has not been declared
   math::Vector3 euler_angles = W_pose_W_C.rot.GetAsEuler();
   ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:171:16: error: ‘euler_angles’ was not declared in this scope
   double phi = euler_angles.x;   // roll
                ^~~~~~~~~~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:171:16: note: suggested alternative: ‘file_handle’
   double phi = euler_angles.x;   // roll
                ^~~~~~~~~~~~
                file_handle
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:175:3: error: ‘math’ has not been declared
   math::Vector3 C_linear_velocity_W_C = link_->GetRelativeLinearVel();
   ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:176:14: error: ‘C_linear_velocity_W_C’ was not declared in this scope
   double u = C_linear_velocity_W_C.x;
              ^~~~~~~~~~~~~~~~~~~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:179:3: error: ‘math’ has not been declared
   math::Vector3 C_angular_velocity_W_C = link_->GetRelativeAngularVel();
   ^~~~
/home/arsalan/in_ws/src/ibvs_sim/plugins/src/boat_forces_and_moments.cpp:180:14: error: ‘C_angular_velocity_W_C’ was not declared in this scope
   double p = C_angular_velocity_W_C.x;
              ^~~~~~~~~~~~~~~~~~~~~~
ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/build.make:62: recipe for target 'ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/plugins/src/boat_forces_and_moments.cpp.o' failed
make[2]: *** [ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/plugins/src/boat_forces_and_moments.cpp.o] Error 1
CMakeFiles/Makefile2:8938: recipe for target 'ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/all' failed
make[1]: *** [ibvs_sim/CMakeFiles/boat_forces_and_moments_plugin.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
can anyone kindly help me to address this issue, thanks

BillThePlatypus, 2019-11-21:
It looks like ibvs_sim is not updated to support Gazebo 8 as a part of ROS Melodic.
This doesn't seem to be a ROSflight error, you'd be best to ask the ibvs_sim maintainers.

closed 2019-11-21 by BillThePlatypus

Arsalan66, 2019-11-22:
Yes I tried ros flight and it ran fine, I tried ibvs sim on ros kinetic as well but it failed that's why I moved on melodic anyways thanks for kind response, sir