rosflight/rosflight_ros_pkgs #154: arm64 support

bsutherland333, 2023-10-16:
rosflight_ros_pkgs currently can't be compiled on arm64 devices, due to Gazebo not having an arm64 target. Enabling some way to still be able to use the functionality that should exist on arm64 devices (rosflight_io, rosflight_utils) while still following the same compilation procedure would make things much simpler on arm64 devices.

JMoore5353, 2024-03-08:
I put compilation checks in the CMakeLists of rosplane_sim and rosflight_sim packages to skip compilation if the architecture is arm64. I also updated our Dockerfile to skip the rosdep keys related to gazebo if the architecture is arm64.

bsutherland333, 2024-03-08:
Could you also update the webpage documentation?

closed 2024-03-11 by JMoore5353