rosflight/rosflight_ros_pkgs #171: Failed to build

AayushAgrawal2003, 2024-07-03:
Ubuntu 22.04 ROS - Humble

When I run 
git clone --recursive https://github.com/rosflight/rosflight.git

I get the following error
fatal: repository 'https://github.com/rosflight/varmint.git/' not found
fatal: clone of 'https://github.com/rosflight/varmint.git' into submodule path '/Users/aayushagrawal/Desktop/Projects/trash/rosflight_ros_pkgs/rosflight_sim/include/rosflight_sim/rosflight_firmware/boards/varmint' failed
Failed to clone 'boards/varmint' a second time, aborting
Failed to recurse into submodule path 'rosflight_sim/include/rosflight_sim/rosflight_firmware'

When I try to build using colcon, it throws a v1.0/rosflight/mavlink.h no such a file or dir error

Thanks!

AayushAgrawal2003, 2024-07-04:
This issue was resolved after the last commit. However, now I am facing a new problem when launching the sim.

[INFO] [launch]: All log files are below ~
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'visualize.rviz' is not a valid package name

'visualize.rviz' is a package, confirmed in ros2 pkg list. Any idea why this issue might be arising?

JMoore5353, 2024-07-18:
Hi @AayushAgrawal2003,

I'm not sure what is causing the issue. Does this happen when you launch the `rosflight_sim` launch files (`fixedwing.launch.py`, `multirotor.launch.py`, ... etc.)?

bsutherland333, 2024-08-15:
This was my bad, I broke the rule of getting rid of code that has been officially 'published'. The ROSflight sim, for a moment, was using a firmware commit that used the old varmint repository instead of the new varmint_h7 repository as a submodule. I deleted the varmint repository and updated the firmware to use the new repository, but didn't update the submodule that the rosflight sim was using.

closed 2024-08-15 by bsutherland333