rosflight/rosflight_ros_pkgs #175: RC node in sim only supports 7 channels

JMoore5353, 2024-07-18:
The RC node in simulation only supports 7 channels (due to a `pygame` limitation). It would be useful to have the same number of RC channels available in simulation as there are on the actual hardware.

iandareid, 2024-08-05:
Pygame seems to have multiple issues, in addition to the 7 channel problem, VimFly  needs a window to accept input, but the GUI window closes when trying to use in conjunction with ROS timers. The work around is very hacky, and makes me uncomfortable.

It seems like we should stop using pygame entirely, I think we can get the same functionality with rqt, and keep everything in the ROS ecosystem. Thoughts? @JMoore5353 @bsutherland333

JMoore5353, 2025-07-18:
I agree the implementation is a bit hacky with a `while` loop in the constructor :) Not sure the best way to fix it given the GUI issues you were talking about.

I think this is a good issue for a newcomer to address. I am ok with getting rid of PyGame, as long as we can have an easy-to-use RC/joystick module that also supports VimFly. Piglet looks like it might be a good option, but I haven't dived into it.