rosflight/rosflight_ros_pkgs #183: Swap Navsat message types with GNSS full.

iandareid, 2024-08-15:
The info in the GNSS full message has everything you would need for an estimator rosflight_io seemingly breaks this into two messages that don't contain all of the info you need, (current date and time). 

I think we should take out the navsat messages from rosflight_io, unless anyone knows why they were in there in the first place. We should have rosflight_io just package up gnss_full and publish it. 

I think this will help with simplicity, and then we can finally get rid of those duplicate GPS messages getting sent from the firmware. @JMoore5353 @bsutherland333 Thoughts?

bsutherland333, 2024-08-15:
The GNSS full message has more information than we really need, but does have some useful things that GNSS does not. I'd propose getting rid of navsat, moving anything from GNSS full that we want into GNSS, and then gettting rid of GNSS full too.

iandareid, 2025-04-28:
This issue will be fixed with an upcoming branch merger where all GNSS information is consolidated into one message.